lwfinger/rtw89 #379: Module does not load in kernel 5.15,  Jetpack 6.2 on Jetson Orin Nano

6thday, 2025-03-13:
Jetson Orin Nano Super + Jetpack 6.2 b77 + RTL8922AE
Not working.....I use latest RTW89 and did clean make before install instructions

(Problem during make)
$ make
/home/keihofung/rtw89/regd.c:724:13: warning: ‘__rtw89_reg_6ghz_power_recalc’ defined but not used [-Wunused-function]
  724 | static void __rtw89_reg_6ghz_power_recalc(struct rtw89_dev *rtwdev)
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

$ uname -r
5.15.148-tegra

$ uname -a
Linux C000 5.15.148-tegra #1 SMP PREEMPT Tue Jan 7 17:14:38 PST 2025 aarch64 aarch64 aarch64 GNU/Linux

$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description: Ubuntu 22.04.5 LTS
Release: 22.04
Codename: jammy

$ lsusb
Bus 002 Device 002: ID 0bda:0489 Realtek Semiconductor Corp. 4-Port USB 3.0 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0bda:8922 Realtek Semiconductor Corp. Bluetooth Radio
Bus 001 Device 002: ID 0bda:5489 Realtek Semiconductor Corp. 4-Port USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

$ sudo dmesg |grep rtw
[   10.702791] rtw89core: module verification failed: signature and/or required key missing - tainting kernel
[   10.789070] rtw89_8922ae 0001:01:00.0: Adding to iommu group 4
[   10.796255] rtw89_8922ae 0001:01:00.0: Direct firmware load for rtw89/rtw8922a_fw.bin failed with error -2
[   10.796295] rtw89_8922ae 0001:01:00.0: Falling back to sysfs fallback for: rtw89/rtw8922a_fw.bin
[   73.042497] rtw89_8922ae 0001:01:00.0: failed to early request firmware: -110
[   73.042695] rtw89_8922ae 0001:01:00.0: enabling device (0000 -> 0003)
[   73.045209] rtw89_8922ae 0001:01:00.0: Direct firmware load for rtw89/rtw8922a_fw.bin failed with error -2
[   73.045218] rtw89_8922ae 0001:01:00.0: Falling back to sysfs fallback for: rtw89/rtw8922a_fw.bin
[  134.475612] rtw89_8922ae 0001:01:00.0: failed to wait firmware completion
[  134.475623] rtw89_8922ae 0001:01:00.0: failed to setup chip information
[  134.478718] rtw89_8922ae: probe of 0001:01:00.0 failed with error -22

a5a5aa555oo, 2025-03-13:
`[ 10.796255] rtw89_8922ae 0001:01:00.0: Direct firmware load for rtw89/rtw8922a_fw.bin failed with error -2`

I am not sure if this driver works on a system with kernel 5.15, but obviously it looks like your system has no firmware needed.

You can run this command to download it from [linux-firmware](https://web.git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtw89)

```
$ wget https://web.git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/rtw89/rtw8922a_fw.bin
```
sha1sum should be `59853a94d0b4a404cd0428800bd27ae4b4a9782a`
```
$ sha1sum  rtw8922a_fw.bin 
59853a94d0b4a404cd0428800bd27ae4b4a9782a  rtw8922a_fw.bin
```

then copy `rtw8922a_fw.bin` to `/lib/firmware/rtw89`
`sudo cp -v rtw8922a_fw.bin /lib/firmware/rtw89`

reboot or reload the rtw89 driver to see if the driver works.
`sudo rmmod rtw_8922ae rtw_8922a rtw89pci rtw89core`
`sudo modprobe rtw_8922ae`

6thday, 2025-03-13:
> `[ 10.796255] rtw89_8922ae 0001:01:00.0: Direct firmware load for rtw89/rtw8922a_fw.bin failed with error -2`
> 
> I am not sure if this driver works on a system with kernel 5.15, but obviously your system lacks of the firmware needed.
> 
> You can run this command to download it from [linux-firmware](https://web.git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtw89)
> 
> ```
> $ wget https://web.git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/rtw89/rtw8922a_fw.bin
> ```
> 
> sha1sum should be `59853a94d0b4a404cd0428800bd27ae4b4a9782a`
> 
> ```
> $ sha1sum  rtw8922a_fw.bin 
> 59853a94d0b4a404cd0428800bd27ae4b4a9782a  rtw8922a_fw.bin
> ```
> 
> then copy `rtw8922a_fw.bin` to `/lib/firmware/rtw89` `sudo cp -v rtw8922a_fw.bin /lib/firmware/rtw89`
> 
> reboot or reload the rtw89 driver to see if the driver works. `sudo rmmod rtw_8922ae rtw_8922a rtw89pci rtw89core` `sudo modprobe rtw_8922ae`

thank you!.. then the result below........

$ sudo dmesg |grep rtw
[   10.124464] rtw89core: module verification failed: signature and/or required key missing - tainting kernel
[   10.158002] rtw89_8922ae 0001:01:00.0: Adding to iommu group 4
[   10.161458] rtw89_8922ae 0001:01:00.0: loaded firmware rtw89/rtw8922a_fw.bin
[   10.161662] rtw89_8922ae 0001:01:00.0: enabling device (0000 -> 0003)
[   10.163576] rtw89_8922ae 0001:01:00.0: no suitable firmware found
[   10.163583] rtw89_8922ae 0001:01:00.0: failed to recognize firmware
[   10.163585] rtw89_8922ae 0001:01:00.0: failed to setup chip information
[   10.166410] rtw89_8922ae: probe of 0001:01:00.0 failed with error -2

a5a5aa555oo, 2025-03-13:
Is the size of the firmware correct?
```
$ /bin/ls -l /lib/firmware/rtw89/rtw8922a_fw.bin 
-rw-r--r-- 1 root root 1537690 Mar 13 20:17 /lib/firmware/rtw89/rtw8922a_fw.bin
```

I took a look at code, but I don't know what is wrong.

btw, I just installed the rtw89 driver here in my Arch Linux (kernel: 5.15.178-1-lts515), but it's didn't work with RTL8852BE.
when I ran `sudo modprobe rtw_8852be`, a network interface doesn't show up and my system almost froze.

6thday, 2025-03-13:
> Is the size of the firmware correct?
>
> ```
> $ /bin/ls -l /lib/firmware/rtw89/rtw8922a_fw.bin
> -rw-r--r-- 1 root root 1537690 Mar 13 20:17 /lib/firmware/rtw89/rtw8922a_fw.bin
> ```
>
> I took a look at [...]

The size is correct...same as urs.... I think the driver is not loaded

$ sudo lshw -C network
  *-network UNCLAIMED       
       description: Network controller
       product: Realtek Semiconductor Co., Ltd.
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0001:01:00.0
       version: 01
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress cap_list
       configuration: latency=0
       resources: ioport:1000(size=256) memory:20a8000000-20a80fffff

a5a5aa555oo, 2025-03-13:
> The size is correct...same as urs.... I think the driver is not loaded
>
> $ sudo lshw -C network *-network UNCLAIMED description: Network controller product: Realtek Semiconductor Co., Ltd. vendor: [...]

What is the output of `lsmod | grep rtw`? Did you remove my rtw89 driver installed a few days ago?

6thday, 2025-03-13:
> What is the output of `lsmod | grep rtw`? Did you remove my rtw89 driver installed a few days ago?

Yes, I did a very clean uninstall from your instructions...

cd rtw89
sudo dkms status
sudo dkms remove rtw89/6.6.66 --all
sudo make uninstall
sudo make uninstall install_fw

------------------

$ lsmod | grep rtw
rtw_8922ae             16384  0
rtw_8922a              61440  1 rtw_8922ae
rtw89pci               73728  1 rtw_8922ae
rtw89core             626688  2 rtw_8922a,rtw89pci
mac80211              778240  2 rtw89pci,rtw89core
cfg80211              765952  3 rtw_8922a,rtw89core,mac80211

a5a5aa555oo, 2025-03-13:
If you see `rtw_8922ae`, `rtw_8922a`, `rtw89pci` and `rtw89core` when running `lsmod`, that means the driver is loaded.
I looked at code again, still didn't know why firmware was not recognized by the driver..

6thday, 2025-03-14:
> If you see `rtw_8922ae`, `rtw_8922a`, `rtw89pci` and `rtw89core` when running `lsmod`, that means the driver is loaded. I looked at code again, still didn't know why firmware was not recognized by the driver..

1)Could it be because the kernel name is different? 5.15.148-tegra

2)I also tried this driver for my RTL8832CU USB but I uninstalled using their script before trying rtw89, I am not sure if it is related

https://github.com/morrownr/rtl8852cu-20240510

Now there is always a *-network I dont know what it is.... I am not sure if it is from the USB-C etc

$ sudo lshw -C network
  *-network UNCLAIMED       
       description: Network controller
       product: Realtek Semiconductor Co., Ltd.
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0001:01:00.0
       version: 01
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress cap_list
       configuration: latency=0
       resources: ioport:1000(size=256) memory:20a8000000-20a80fffff
  *-network
       description: Ethernet interface
       product: RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
       vendor: Realtek Semiconductor Co., Ltd.
       physical id: 0
       bus info: pci@0008:01:00.0
       logical name: enP8p1s0
       version: 15
       serial: 48:b0:2d:f7:55:16
       size: 1Gbit/s
       capacity: 1Gbit/s
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress msix vpd bus_master cap_list ethernet physical tp 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=r8168 driverversion=8.053.00-NAPI duplex=full ip=192.168.21.167 latency=0 link=yes multicast=yes port=twisted pair speed=1Gbit/s
       resources: irq:189 ioport:300000(size=256) memory:3528004000-3528004fff memory:3528000000-3528003fff
  *-network
       description: Ethernet interface
       physical id: c
       logical name: usb0
       serial: b2:57:20:9f:58:cd
       capabilities: ethernet physical
       configuration: broadcast=yes driver=g_ether driverversion=29-May-2008 firmware=tegra-xudc link=no multicast=yes