lwfinger/rtw89 #380: Driver Not Functioning: HP 255 G10 (Arch Linux) - FW Does Not Process H2C Registers

NLion74, 2025-04-02:
**Description**
Initially, the rtw89 driver worked for a short time (0-30 minutes) after boot but would eventually crash. Recently, it has stopped working entirely.

**System Information**

- Laptop: HP 255 G10 853S9ES
- Kernel: 6.13.8-arch1-1
- Driver Version: Latest from rtw89-dkms-git AUR and lwfinger’s GitHub repository

**Steps to Reproduce**
1. Install the driver from the Aur
```
git clone https://aur.archlinux.org/rtw89-dkms-git.git
cd rtw89-dkms-git
makepkg -sri
cd ..
```
2. Apply the configuration
```
git clone https://github.com/lwfinger/rtw89.git
cd rtw89
sudo cp 70-rtw89.conf /etc/modprobe.d/
sudo cp suspend_rtw89 /usr/lib/systemd/system-sleep/
reboot
```
3.  After the reboot, the driver works for a short time period.

Logs of `sudo dmesg`:
```
[   45.352694] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.352782] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.352879] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.352976] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.353073] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.353161] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.353258] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.353357] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.353456] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.353553] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.353641] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.353738] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.353835] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.353906] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.353970] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.354033] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.354096] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.354158] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.354221] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.354283] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.354348] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.354411] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.354473] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.354536] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.354598] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.354661] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.354724] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.354786] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.354849] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.354911] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.354974] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.355036] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.355099] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.355162] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.355224] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.355287] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.355353] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.355416] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.355478] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.355541] rtw89_8852be 0000:01:00.0: read rf busy swsi
[   45.355603] rtw89_8852be 0000:01:00.0: read rf busy swsi
[... 155 lines cut ...]
[   45.372302] rtw89_8852be 0000:01:00.0: timed out to flush pci txch: 9
[   45.424651] rtw89_8852be 0000:01:00.0: xtal si not ready(W): offset=90 val=10 mask=10
[   45.424933] rtw89_8852be 0000:01:00.0: timed out to flush pci txch: 0
[   45.425057] rtw89_8852be 0000:01:00.0: timed out to flush pci txch: 1
[   45.425180] rtw89_8852be 0000:01:00.0: timed out to flush pci txch: 2
[   45.425304] rtw89_8852be 0000:01:00.0: timed out to flush pci txch: 3
[   45.425435] rtw89_8852be 0000:01:00.0: timed out to flush pci txch: 8
[   45.425558] rtw89_8852be 0000:01:00.0: timed out to flush pci txch: 9
[   45.425562] rtw89_8852be 0000:01:00.0: failed to pre-release fwcmd
[   46.999389] rtw89_8852be 0000:01:00.0: xtal si not ready(W): offset=90 val=10 mask=10
[   47.020334] rtw89_8852be 0000:01:00.0: mac init fail, ret:-110
[   47.091381] rtw89_8852be 0000:01:00.0: xtal si not ready(W): offset=90 val=10 mask=10
[   47.112333] rtw89_8852be 0000:01:00.0: mac init fail, ret:-110
[   48.767387] rtw89_8852be 0000:01:00.0: xtal si not ready(W): offset=90 val=10 mask=10
[   48.788336] rtw89_8852be 0000:01:00.0: mac init fail, ret:-110
[   48.859383] rtw89_8852be 0000:01:00.0: xtal si not ready(W): offset=90 val=10 mask=10
[   48.880367] rtw89_8852be 0000:01:00.0: mac init fail, ret:-110
[   52.943388] rtw89_8852be 0000:01:00.0: xtal si not ready(W): offset=90 val=10 mask=10
[   52.964333] rtw89_8852be 0000:01:00.0: mac init fail, ret:-110
[   53.035387] rtw89_8852be 0000:01:00.0: xtal si not ready(W): offset=90 val=10 mask=10
[   53.056336] rtw89_8852be 0000:01:00.0: mac init fail, ret:-110

```

I have tried to reinstall the driver countless time. The issue seems to persist between reboots. Let me know if you need any additional logs,

claraphyll, 2025-04-03:
Same laptop model here, this configuration change fixed it for me: <https://wiki.archlinux.org/title/HP_255_G10#Wi-Fi>

Although speed is still worse than on a comparably priced ThinkPad

NLion74, 2025-04-03:
Thanks, I applied the following in `/etc/modprobe.d/rtw89-fix.conf`

```
rtw89_pci disable_aspm_l1=Y disable_aspm_l1ss=Y
rtw89_core disable_ps_mode=y
```

For now it seems to work which really suprised me. I won't close the issue just yet as I had the exact same scenario only 2 days ago where it worked for 2 hours but then crashed.

NLion74, 2025-04-03:
It does seem to have fixed my issue

NLion74, 2025-04-03:
I have run into the same issue once again with the following error when running `sudo dmesg`
```[ 1418.405969] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.406028] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.406089] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.406150] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.406209] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.406270] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.406335] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.406396] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.406456] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.406515] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.406575] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.406634] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.406694] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.406754] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.406814] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.406874] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.406934] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.406994] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.407056] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.407118] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.407178] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.407238] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.407301] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.407365] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.407426] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.407485] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.407545] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.407605] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.407665] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.407724] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.407784] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.407843] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.407903] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.407962] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.408022] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.408081] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.408141] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.408200] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.408260] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.408322] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.408384] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.408444] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.408504] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.408563] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.408623] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.408686] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.408748] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.408811] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.408874] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.408936] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.408999] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.409062] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.409125] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.409189] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.409251] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.409318] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.409382] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.409445] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.409507] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.409570] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.409633] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.409695] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.409758] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.409821] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.409883] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.409946] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.410008] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.410071] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1418.410134] rtw89_8852be 0000:01:00.0: read rf busy swsi
[... 231 lines cut ...]
[ 1428.925055] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1428.925122] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1430.919815] rtw89_8852be 0000:01:00.0: firmware failed to ack for leaving ps mode
[ 1430.970130] rtw89_8852be 0000:01:00.0: rtw89: failed to leave lps state
[ 1430.970251] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.970356] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.970449] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.970537] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.970626] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.970718] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.970806] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.970894] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.970993] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.971082] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.971177] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.971273] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.971343] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.971408] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.971473] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.971539] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.971604] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.971669] rtw89_8852be 0000:01:00.0: write rf busy swsi
[ 1430.973045] rtw89_8852be 0000:01:00.0: read rf busy swsi
[ 1430.973118] rtw89_8852be 0000:01:00.0: read rf busy swsi
```

claraphyll, 2025-04-03:
The file you posted doesn't apply the options. You have to add `options` in front of every line. https://wiki.archlinux.org/title/Kernel_module#Using_modprobe.d

NLion74, 2025-04-03:
Oh yes apologies I updated the files but it still doesnt work. Now the driver stopped working completely, the wifi interface won't even show anymore. I have rebuilt the initramfs with `sudo mkinitcpio -P`

My current modprobe configurations are as follows:
```
cat /etc/modprobe.d/70-rtw89.conf 

# set options for faulty HP and Lenovo BIOS code
options rtw89_pci disable_aspm_l1=y disable_aspm_l1ss
options rtw89pci disable_aspm_l1=y disable_aspm_l1ss

cat /etc/modprobe.d/rtw89-fix.conf 

options rtw89_core disable_ps_mode=y

cat /etc/modprobe.d/blacklist-rtw89.conf 

blacklist rtw89_8851be
blacklist rtw89_8851b
blacklist rtw89_8852ae
blacklist rtw89_8852a
blacklist rtw89_8852b_common
blacklist rtw89_8852be
blacklist rtw89_8852b
blacklist rtw89_8852bte
blacklist rtw89_8852bt
blacklist rtw89_8852ce
blacklist rtw89_8852c
blacklist rtw89_8922ae
blacklist rtw89_8922a
blacklist rtw89_core
blacklist rtw89_pci
```

I am not sure what to try next. I could try to disable fast boot like suggested in the guide, I however couldn't find the option in the bios so far-

claraphyll, 2025-04-03:
Then I don't know. I remember having to shutdown and power up or restart the laptop a couple of times for the card to reappear.

I never used the blacklist or this AUR package, the blacklist would probably make the card not appear at all.

Fast boot is only relevant if you dual-boot with Windows but after a "cold boot" it should not be broken any longer

NLion74, 2025-04-03:
Yes I assume the blacklist also wouldn't be applied as there is no `options` infront of each line. I have rebooted the laptop many times but it does not seem to work unfortunately. Thank you for your help though.

claraphyll, 2025-04-03:
No, the blacklist will parse and you should remove it. "blacklist" is the command for each line in the same way "options" is the command in the correct file.

NLion74, 2025-04-03:
I understand. I do believe I may need the blacklist though to prevent driver issues with the linux-firmware default one from conflicting with the aur installed one.

claraphyll, 2025-04-03:
Maybe try 
- *with* vanilla kernel
- *without* the AUR package
- *with* options file
- *without* blacklist

first. That's what I got to work consistently (for several days with only suspend in between) for me (although limited to about 100 MBit/s).

I have to go do some work now, good luck

NLion74, 2025-04-03:
@a5a5aa555oo Thanks for now this seems to work as for now. 

Running 
```
cat /sys/module/rtw89/parameters/disable_aspm_l1
cat /sys/module/rtw89/parameters/disable_aspm_l1ss
cat /sys/module/rtw89/parameters/disable_ps_mode
```

outputs 
```
Y
Y
Y
```

I'll continue using my laptop as normal and see if something ends up breaking.

NLion74, 2025-04-04:
This does seem to have finally fixed my issue, thank you @a5a5aa555oo

NLion74, 2025-05-08:
Hey @a5a5aa555oo,

thanks for the tip. I updated my driver to the latest version and adjust the config like you said. It works flawlessly and enabled me to remove my other custom configs.

The commands output the following which seems fine too
```
cat /sys/module/rtw89_core_git/parameters/disable_ps_mode
cat /sys/module/rtw89_pci_git/parameters/disable_aspm_l1
cat /sys/module/rtw89_pci_git/parameters/disable_aspm_l1ss
Y
Y
Y
```

Thanks again!

NLion74, 2025-05-17:
Unfortunately I have to reopen this issue. I had no problems for a couple of days even after applying what @a5a5aa555oo suggested. Now suddenly it stopped working again. I changed the options in modprobe config and I do get this output

```
cat /sys/module/rtw89_core_git/parameters/disable_ps_mode
cat /sys/module/rtw89_pci_git/parameters/disable_aspm_l1
cat /sys/module/rtw89_pci_git/parameters/disable_aspm_l1ss
Y
Y
Y
```

dmesg shows the following output.

```
[  114.857701] rtw89_8852be_git 0000:01:00.0: failed to pre-release fwcmd
[  114.857819] rtw89_8852be_git 0000:01:00.0: read rf busy swsi
[  114.857889] rtw89_8852be_git 0000:01:00.0: read rf busy swsi
[  116.834865] rtw89_8852be_git 0000:01:00.0: read rf busy swsi
[  116.834944] rtw89_8852be_git 0000:01:00.0: read rf busy swsi
[  118.930630] rtw89_8852be_git 0000:01:00.0: rtw89: failed to leave lps state
[  118.930779] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.930901] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.931005] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.931107] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.931209] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.931319] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.931418] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.931487] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.931553] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.931622] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.931692] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.931758] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.931827] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.931896] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.931965] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.932032] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.932099] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.932165] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  118.933336] rtw89_8852be_git 0000:01:00.0: read rf busy swsi
[  118.933407] rtw89_8852be_git 0000:01:00.0: read rf busy swsi
[  120.915195] rtw89_8852be_git 0000:01:00.0: rtw89: failed to leave lps state
[  120.915282] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.915353] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.915421] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.915488] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.915555] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.915622] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.915690] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.915757] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.915824] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.915891] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.915958] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.916025] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.916114] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.916185] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.916253] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.916320] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.916387] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.916454] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  120.917816] rtw89_8852be_git 0000:01:00.0: read rf busy swsi
[  120.917888] rtw89_8852be_git 0000:01:00.0: read rf busy swsi
[  122.899070] rtw89_8852be_git 0000:01:00.0: rtw89: failed to leave lps state
[  122.899158] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  122.899229] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  122.899297] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  122.899365] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  122.899433] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  122.899501] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  122.899569] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  122.899638] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[... 15 lines cut ...]
[  124.882185] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  124.882256] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  124.882326] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  124.882397] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  124.882467] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  124.882538] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  124.882608] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  124.882686] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  124.882749] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  124.882811] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  124.882874] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  124.882937] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  124.883000] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  124.883062] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  124.883125] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  124.883188] rtw89_8852be_git 0000:01:00.0: write rf busy swsi
[  124.884556] rtw89_8852be_git 0000:01:00.0: read rf busy swsi
[  124.884620] rtw89_8852be_git 0000:01:00.0: read rf busy swsi
```

I don't know what changed, this didn't happen immeaditly after an update of the kernel or anything else but seemingly at random

My current kernel version is `6.14.6-arch1-1`

If you need any further logs or have any other questions feel free to ask.

a5a5aa555oo, 2025-05-17:
@NLion74 

Which version of rtw89-dkms-git are you using now?

NLion74, 2025-05-17:
@a5a5aa555oo Thanks for the quick reply. According to my paru aur package manager my currently installed version is `r1181.73630b6-1`

a5a5aa555oo, 2025-05-17:
Not sure what is the cause, I would recommend you to try the built-in rtw89 driver in kernel 6.14 which may be more stable.

Steps:
1. Remove rtw89-dkms-git
```
sudo pacman -Rn rtw89-dkms-git
```

2. Create `rtw89.conf` in `/etc/modprobe.d/` with the following lines
```
options rtw89_core disable_ps_mode=y
options rtw89_pci disable_aspm_l1ss=y 
options rtw89_pci disable_aspm_l1=y
```

Then, shutdown and cold boot your laptop.

NLion74, 2025-05-17:
@a5a5aa555oo This seems to have worked. Not sure what abt the AUR Version caused this but the kernel version is fully functional. Thanks again!

NLion74, 2025-05-24:
So at this it honestly seems kinda funny but yes, you guessed it. It stopped working again. I still have my config with

```
cat /etc/modprobe.d/70-rtw89.conf

options rtw89_core disable_ps_mode=y
options rtw_pcie disable_aspm_l1ss=y
options rtw_pcie disable_aspm_l1=y
```

And the dmesg error is the same again:
```
[  387.623464] rtw89_8852be 0000:01:00.0: rtw89: failed to leave lps state
[  387.623576] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.623659] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.623738] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.623827] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.623907] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.623985] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.624065] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.624143] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.624222] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.624300] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.624379] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.624457] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.624536] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.624615] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.624690] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.624761] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.624837] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.624910] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  387.626273] rtw89_8852be 0000:01:00.0: read rf busy swsi
[  387.626343] rtw89_8852be 0000:01:00.0: read rf busy swsi
[  389.607988] rtw89_8852be 0000:01:00.0: rtw89: failed to leave lps state
[  389.608098] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.608176] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.608251] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.608333] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.608408] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.608482] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.608556] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.608631] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.608705] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.608779] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.608853] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.608927] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.609001] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.609076] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.609150] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.609224] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.609302] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.609377] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  389.610750] rtw89_8852be 0000:01:00.0: read rf busy swsi
[  389.610825] rtw89_8852be 0000:01:00.0: read rf busy swsi
[  391.592431] rtw89_8852be 0000:01:00.0: rtw89: failed to leave lps state
[  391.592547] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  391.592625] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  391.592701] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  391.592783] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  391.592858] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  391.592932] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  391.593006] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  391.593080] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  391.593155] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  391.593229] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  391.593302] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  391.593372] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  391.593443] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  391.593514] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  391.593585] rtw89_8852be 0000:01:00.0: write rf busy swsi
[... 25 lines cut ...]
[  393.579690] rtw89_8852be 0000:01:00.0: read rf busy swsi
[  395.625113] rtw89_8852be 0000:01:00.0: rtw89: failed to leave lps state
[  395.625205] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.625279] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.625350] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.625421] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.625491] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.625562] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.625640] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.625710] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.625780] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.625851] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.625921] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.625991] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.626058] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.626125] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.626191] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.626258] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.626324] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.626391] rtw89_8852be 0000:01:00.0: write rf busy swsi
[  395.627734] rtw89_8852be 0000:01:00.0: read rf busy swsi
[  395.627804] rtw89_8852be 0000:01:00.0: read rf busy swsi
```

And just like last time this happend after an update. I really don't know why this broke or how to fix this long term

a5a5aa555oo, 2025-05-24:
Hi,

There is something strange in your 70-rtw89.conf, the module names of the rtw89 driver in the stock kernel are "rtw89_core" and "rtw89_pci", so the correct settings should be:

```
options rtw89_core disable_ps_mode=y
options rtw89_pci disable_aspm_l1ss=y
options rtw89_pci disable_aspm_l1=y
```

You can run these commands to check to value of these module parameters.
```
cat /sys/module/rtw89_pci/parameters/disable_aspm_l1
cat /sys/module/rtw89_pci/parameters/disable_aspm_l1ss
cat /sys/module/rtw89_core/parameters/disable_ps_mode
```

[image]

NLion74, 2025-05-24:
Hey @a5a5aa555oo, I didnt realise these options were wrong. I changed them and rebooted

It now looks as follows:
```
cat /etc/modprobe.d/70-rtw89.conf

options rtw89_core disable_ps_mode=y
options rtw89_pci disable_aspm_l1ss=y
options rtw89_pci disable_aspm_l1=y
```

However the according sys modules don't seem to have appeared:
```
cat /sys/module/rtw89_pci/parameters/disable_aspm_l1
cat /sys/module/rtw89_pci/parameters/disable_aspm_l1ss
cat /sys/module/rtw89_core/parameters/disable_ps_mode

cat: /sys/module/rtw89_pci/parameters/disable_aspm_l1: No such file or directory
cat: /sys/module/rtw89_pci/parameters/disable_aspm_l1ss: No such file or directory
Y
```

A new behaviour I observed is that my wifi interface has disappeared completely. This has never happend before, usually it still showed up but didn't work:
```
ip link show

1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
```

Feel free to ask for any other command outputs or logs :)

a5a5aa555oo, 2025-05-24:
Here are some useful tips for you.

1. How can I know the module names of the driver i am using?

Run `lsmod | grep rtw` and you should see "rtw89_8852be", "rtw89_8852b", "rtw89_pci" and "rtw89_core" are loaded to drive your 8852BE device.
```
$ lsmod | grep rtw
rtw89_8852be           12288  0
rtw89_8852b           376832  1 rtw89_8852be
rtw89_pci              86016  1 rtw89_8852be
rtw89_core            663552  2 rtw89_8852b,rtw89_pci
mac80211             1568768  2 rtw89_core,rtw89_pci
cfg80211             1331200  3 rtw89_8852b,rtw89_core,mac80211
```

2. How can I know the module parameter available for each module?

Run `modinfo YOUR_MODULE_NAME | grep parm`

```
$ modinfo rtw89_core | grep parm
parm:           debug_mask:Debugging mask (uint)
parm:           disable_ps_mode:Set Y to disable low power mode (bool)

$ modinfo rtw89_pci | grep parm
parm:           disable_clkreq:Set Y to disable PCI clkreq support (bool)
parm:           disable_aspm_l1:Set Y to disable PCI ASPM L1 support (bool)
parm:           disable_aspm_l1ss:Set Y to disable PCI L1SS support (bool)

$ modinfo rtw89_8852b | grep parm
$ modinfo rtw89_8852be | grep parm

(rtw89_8852b and rtw89_8852be don't have any module parameter.)
```

3. How to set the module parameter for the specific module?

Create a `xxxx.conf` (naming doesn't matter.) in `/etc/modprobe.d` with the following lines

```
# The first field must be "options"
# Fill the module name in the second field
# Fill module parameter and its value in the third field

options MODULE_NAME MODULE_PARAMETER=VALUE
```

a5a5aa555oo, 2025-05-24:
@NLion74 

Could you post the output of this command? it will find out all the out-of-kernel rtw89 driver installed in your system.
```
sudo find /lib/modules/$(uname -r) -name "rtw_*"
```

NLion74, 2025-05-24:
Yes of course, however the output was empty

a5a5aa555oo, 2025-05-24:
Could you attach the full output of dmesg?
```
sudo dmesg > dmesg.txt
```

NLion74, 2025-05-24:
Sure here's the attachment:

[dmesg.txt](https://github.com/user-attachments/files/20427595/dmesg.txt)

a5a5aa555oo, 2025-05-24:
Does it help if you cold boot your laptop?
You must shutdown your laptop completely and startup it again after a few seconds, don't just "reboot" it.

NLion74, 2025-05-24:
Okay I think that actually fixed it for now.

```
Lsmod | grep rtw

rtw89_8852be           12288  0
rtw89_8852b           339968  1 rtw89_8852be
rtw89_8852b_common     65536  1 rtw89_8852b
rtw89_pci             114688  1 rtw89_8852be
rtw89_core           1077248  3 rtw89_8852b,rtw89_pci,rtw89_8852b_common
mac80211             1646592  2 rtw89_core,rtw89_pci
cfg80211             1400832  3 rtw89_core,mac80211,rtw89_8852b_common
rfkill                 40960  9 hp_wmi,rtw89_core,bluetooth,cfg80211

cat /sys/module/rtw89_pci/parameters/disable_aspm_l1
cat /sys/module/rtw89_pci/parameters/disable_aspm_l1ss
cat /sys/module/rtw89_core/parameters/disable_ps_mode

Y
Y
Y
```

I really do not understand why this realtek wifi chip is so unstable. Thanks for rescuing me again, I should give you virtual flowers or something at this point.

a5a5aa555oo, 2025-05-24:
Yes, some Realtek Wi-Fi chips have compatibility problems and are not the best choices for Linux users.
If you are tired of endless debugging, buy an Intel AX200/210 which should have less problems on Linux to replace your 8852BE card.

Fruktsallad04, 2025-09-24:
For anyone late to the party, i had the same problem, but this fix was not enough. What seems to be working for me is both adding the config file mentioned above, AND disabling msi by one of the grub files. i ran `nano `/etc/default/grub` and added `pci=nomsi,noaer` in the comment after `GRUB_CMDFILE_LINUX_DEFAULT=`  
Mistral's le chat gave me these fixes, so if you dont wanna take llm's advice dont do this. 
Im also new to both linux generally and commenting on git, so sorry if the formatting is wrong.