lwfinger/rtw89 #381: Cannot compile on Fedora 41/42

Blizzardo1, 2025-04-04:
root@fedora:/usr/src/rtw89# make
make -C /lib/modules/6.14.0-63.fc42.x86_64/build M=/usr/src/rtw89 modules
make[1]: Entering directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
make[2]: Entering directory '/usr/src/rtw89'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 15.0.1 20250228 (Red Hat 15.0.1-0)
  You are using:           gcc (GCC) 15.0.1 20250329 (Red Hat 15.0.1-0)
  CC [M]  core.o
core.c: In function ‘rtw89_core_napi_init’:
core.c:2591:9: error: implicit declaration of function ‘init_dummy_netdev’ [-Wimplicit-function-declaration]
 2591 |         init_dummy_netdev(&rtwdev->netdev);
      |         ^~~~~~~~~~~~~~~~~
make[4]: *** [/usr/src/kernels/6.14.0-63.fc42.x86_64/scripts/Makefile.build:207: core.o] Error 1
make[3]: *** [/usr/src/kernels/6.14.0-63.fc42.x86_64/Makefile:2040: .] Error 2
make[2]: *** [/usr/src/kernels/6.14.0-63.fc42.x86_64/Makefile:263: __sub-make] Error 2
make[2]: Leaving directory '/usr/src/rtw89'
make[1]: *** [Makefile:263: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.14.0-63.fc42.x86_64'
make: *** [Makefile:104: all] Error 2
root@fedora:/usr/src/rtw89#

koksyn, 2025-04-08:
Same problem on Pop_OS! 22 LTS with Linux kernel 6.12

a5a5aa555oo, 2025-04-09:
@Blizzardo1 @ExtReMLapin @koksyn 

Use the rtw89 driver in your kernel which is much newer than the one here.

The rtw89 driver in this repo = The rtw89 driver in kernel 6.10

Blizzardo1, 2025-04-09:
🤣🤣🤣

Thats the one that's freaking broken. Can't see any Bluetooth device.

a5a5aa555oo, 2025-04-09:
@Blizzardo1 

First, you need to check if the Bluetooth device is detected
`lsusb | grep -i bluetooth`

If your Bluetooth device is detected , check if the driver `btusb` is loaded.
`lsmod | grep btusb`

Blizzardo1, 2025-04-09:
Oh but it is. But it doesn't scan at all

a5a5aa555oo, 2025-04-09:
@Blizzardo1 

What chip do you have? Please run `sudo dmesg > dmesg.txt` and attach the file `dmesg.txt`

Blizzardo1, 2025-04-09:
It's an 8851BE, but that's the thing, it refuses to allocate any findings through bluetoothctl and the dialog.

a5a5aa555oo, 2025-05-26:
@yudin-yaroslav 

What Realtek Wi-Fi card do you have? Could you post the output of these two commands?

```
lspci
```

```
lsusb
```

cmray, 2026-03-20:
I have a the same "implicit declaration" error when compiling under Ubuntu 24.04 with kernel 6.17, on AMD Ryzen 7 pro hardware.   While the drivers included with the Ubuntu distribution did allow the RTL8852BE to connect to the wifi router, the data rate was at most 0.16 Mbps so it was effectively unusable. 

I believe the relevant line of the **lspci** output is 

02:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8852BE PCIe 802.11ax Wireless Network Controller

and for **lsusb**

Bus 001 Device 003: ID 0bda:b85b Realtek Semiconductor Corp. Bluetooth Radio

Any help would be much appreciated.

Blizzardo1, 2026-03-20:
> I have a the same "implicit declaration" error when compiling under Ubuntu 24.04 with kernel 6.17, on AMD Ryzen 7 pro hardware.   While the drivers included with the Ubuntu distribution did allow [...]

Saw your email. No luck, but i haven't tried since. Ill see if i can try again and reproduce the error.