lwfinger/rtw89 #382: Spam error messages with 8922ae

dekar-sun, 2025-04-10:
Hello!
I have problem with my 8922ae card in Opensuse Thumbleweed - periodical spam in the dmesg log:

`rtw89_8922ae 0000:0a:00.0: failed to update 54 RXBD info: -11
`

My network card is 
```
0a:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8922AE 802.11be PCIe Wireless Network Adapter (rev 01)
        Subsystem: Realtek Semiconductor Co., Ltd. RTL8922AE 802.11be PCIe Wireless Network Adapter
        Kernel driver in use: rtw89_8922ae
        Kernel modules: rtw89_8922ae
```

Modules used:
```
lsmod |grep rtw
rtw89_8922ae           12288  0
rtw89_8922a            73728  1 rtw89_8922ae
rtw89_pci             114688  1 rtw89_8922ae
rtw89_core            929792  2 rtw89_8922a,rtw89_pci
mac80211             1519616  2 rtw89_core,rtw89_pci
cfg80211             1409024  3 rtw89_core,rtw89_8922a,mac80211
rfkill                 40960  8 rtw89_core,bluetooth,cfg80211
```
Package version with kernel module:
```
Information for package kernel-firmware-realtek:
------------------------------------------------
Repository     : openSUSE-Tumbleweed-Oss
Name           : kernel-firmware-realtek
Version        : 20250313-1.1
Arch           : noarch
Vendor         : openSUSE
Installed Size : 3.9 MiB
Installed      : Yes
Status         : up-to-date
Source package : kernel-firmware-realtek-20250313-1.1.src
Upstream URL   : https://git.kernel.org/cgit/linux/kernel/git/firmware/linux-firmware.git/
Summary        : Kernel firmware files for Realtek wireless drivers
Description    : 
    This package contains kernel firmware files for Realtek wireless drivers.
```
I've also tried to use the below options but them doesn't help:
`rtw89_core.disable_ps_mode=Y rtw89_pci.disable_clkreq=Y rtw89_pci.disable_aspm_l1=Y rtw89_pci.disable_aspm_l1ss=Y`

a5a5aa555oo, 2025-05-03:
Hi,

As mentioned in README, please file a bug report to linux-wireless (linux-wireless@vger.kernel.org), the developers from Realtek
will see your mail and only they can solve the problem you encountered.

[image]

Examples:
https://lore.kernel.org/linux-wireless/uidltsdsuujrczrtzgerhh5cw2tztxktfen6yvztnc7gttzgvk@jccomj7f4gul/
https://lore.kernel.org/linux-wireless/20240917055730.Horde.pVyI-XF1MRA_zUQ5BDJZ4lC@ignaz.org/

dekar-sun, 2025-05-18:
Ok.
I found that there is already a patch for this behavior: https://yhbt.net/lore/linux-wireless/20250509013433.7573-1-pkshih@realtek.com/#r