lwfinger/rtw89 #384: RTW8922ae "No suitable firmware found" error

Hansan2011, 2025-04-20:
I have a relative new motherboard (ASUS TUF GAMING B850 plus wifi'  with build in wifi with a Realtek 8922 chipset. 
Ubuntu 24.04 with kernel 6.8.0-58-generic #60-Ubuntu does not support this chip yet. 

Therefore, I compiled the driver from this repository following the instructions.  I can install this driver but it has a problem with the firmware:

```shell
hansan@Desk-computer:/usr/src/rtw89-1.0.2$ sudo modprobe -v rtw_8922ae
insmod /lib/modules/6.8.0-58-generic/updates/dkms/rtw89core.ko.zst 
insmod /lib/modules/6.8.0-58-generic/updates/dkms/rtw89pci.ko.zst 
insmod /lib/modules/6.8.0-58-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw_8922a.ko 
insmod /lib/modules/6.8.0-58-generic/kernel/drivers/net/wireless/realtek/rtw89/rtw_8922ae.ko 
``` 
Error message in dmesg:
```shell
[  405.383173] rtw89_8922ae 0000:0a:00.0: loaded firmware rtw89/rtw8922a_fw.bin
[  405.383216] rtw89_8922ae 0000:0a:00.0: enabling device (0000 -> 0003)
[  405.383737] rtw89_8922ae 0000:0a:00.0: no suitable firmware found
[  405.383740] rtw89_8922ae 0000:0a:00.0: failed to recognize firmware
[  405.383741] rtw89_8922ae 0000:0a:00.0: failed to setup chip information
[  405.383918] rtw89_8922ae: probe of 0000:0a:00.0 failed with error -2
```
The firmware is from the repository mentioned in the installation instructions and copied to the (I believe) right location.
```shell
hansan@Desk-computer:~/work/rtw8922ae/rtw89$ ls -l /lib/firmware/rtw89
total 15272
-rw-r--r-- 1 root root 1162024 Apr 19 22:39 rtw8851b_fw.bin
-rw-r--r-- 1 root root 1428272 Apr 19 22:39 rtw8852a_fw.bin
-rw-r--r-- 1 root root 1244784 Apr 19 22:39 rtw8852b_fw-1.bin
-rw-r--r-- 1 root root 1035232 Apr 19 22:39 rtw8852b_fw.bin
-rw-r--r-- 1 root root  918376 Apr 19 22:39 rtw8852bt_fw.bin
-rw-r--r-- 1 root root 2399272 Apr 19 22:39 rtw8852c_fw-1.bin
-rw-r--r-- 1 root root 1532656 Apr 19 22:39 rtw8852c_fw.bin
-rw-r--r-- 1 root root 1583434 Apr 19 22:39 rtw8922a_fw-1.bin
-rw-r--r-- 1 root root 1586954 Apr 19 22:39 rtw8922a_fw-2.bin
-rw-r--r-- 1 root root 1188926 Apr 19 22:39 rtw8922a_fw-3.bin
-rw-r--r-- 1 root root 1537690 Apr 19 22:55 rtw8922a_fw.bin
```
The output of lspci -vk 
```shell
0a:00.0 Network controller: Realtek Semiconductor Co., Ltd. Device 8922 (rev 01)
	Subsystem: AzureWave Device 6011
	Flags: fast devsel, IRQ 38, IOMMU group 19
	I/O ports at c000 [size=256]
	Memory at f6c00000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [40] Power Management version 3
	Capabilities: [50] MSI: Enable- Count=1/8 Maskable+ 64bit+
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [148] Device Serial Number 00-e0-4c-ff-fe-89-22-01
	Capabilities: [158] Latency Tolerance Reporting
	Capabilities: [160] L1 PM Substates
	Capabilities: [170] Vendor Specific Information: ID=0002 Rev=4 Len=100 <?>
	Capabilities: [270] Data Link Feature <?>
	Kernel modules: rtw_8922ae
```

How can I help to resolve the problem?  What do I need to do to debug this problem better?

Hansan2011, 2025-04-20:
To answer my own question.  It does work when I use kernel 6.14.3
```shell
hansan@Desk-computer:/mnt/raid-disk/home2/hansan/rtw8922ae$ uname -a
Linux Desk-computer 6.14.3-061403-generic #202504200838 SMP PREEMPT_DYNAMIC Sun Apr 20 09:12:44 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
hansan@Desk-computer:/mnt/raid-disk/home2/hansan/rtw8922ae$ sudo dmesg | grep rtw
[    5.069593] rtw89_8922ae 0000:0a:00.0: loaded firmware rtw89/rtw8922a_fw-3.bin
[    5.069651] rtw89_8922ae 0000:0a:00.0: enabling device (0000 -> 0003)
[    5.070235] rtw89_8922ae 0000:0a:00.0: Firmware version 0.35.64.0 (c96c445c), cmd version 1, type 1
[    5.070238] rtw89_8922ae 0000:0a:00.0: Firmware version 0.35.64.0 (c96c445c), cmd version 1, type 3
[    5.096551] rtw89_8922ae 0000:0a:00.0: chip rfe_type is 1
[    5.101273] rtw89_8922ae 0000:0a:00.0: Firmware version 0.1.0.0 (7b393818), cmd version 0, type 64
[    5.101278] rtw89_8922ae 0000:0a:00.0: Firmware element BB version: 00 41 00 00
[    5.101283] rtw89_8922ae 0000:0a:00.0: Firmware element radio A version: 00 33 00 00
[    5.101287] rtw89_8922ae 0000:0a:00.0: Firmware element NCTL version: 00 0f 00 00
[    5.101301] rtw89_8922ae 0000:0a:00.0: Firmware element TXPWR version: 00 33 00 00
[    5.101302] rtw89_8922ae 0000:0a:00.0: Firmware element PWR_TRK version: 00 33 00 00
[    5.101945] rtw89_8922ae 0000:0a:00.0: rfkill hardware state changed to enable
[    5.143332] rtw89_8922ae 0000:0a:00.0 wlp10s0: renamed from wlan0
``` 
Therefore, it would be nice to find a solution for kernel 6.8, but that is not really necessary. Ubuntu will also move to 6.14 overtime.

AcidicSkittles, 2025-07-30:
I'm having the same problem

fridec13, 2026-09-01:
**Solution for kernel 6.8 (no kernel upgrade needed)**

For anyone hitting this on a kernel < 6.9, here's what worked for me.

**Hardware:** RTL8922AE (PCI `10ec:8922`) in a Lenovo IdeaPad Slim 5 14AGP11
**OS:** Ubuntu 22.04.5, kernel 6.8.0-138-generic

**Root cause:** newer RTL8922AE units are a newer silicon cut (chip `cv=2` / CCV; dmesg from a working driver shows `chip info CID: 71, CV: 2`). This repo's code only accepts firmware with an exact cv match (`fw.c: mfw_info->cv == rtwdev->hal.cv`), and none of the `rtw8922a_fw*.bin` files in linux-firmware contain a cv=2 image — mainline instead selects the closest cv ≤ chip cv.

I tried backporting mainline's closest-cv selection (both in `rtw89_mfw_recognize()` and in `__rtw89_fw_recognize_from_elm()` for the BBMCU element). That gets the probe to succeed and `wlo1` to appear, but the radio stays deaf: scans find nothing, and with fw-3 the RF calibration fails (`failed to do RF RX_DCK result from state 4`, `HW scan failed with status: -14`). The driver↔firmware H2C/RFK API has moved too much since this repo's last update.

**What actually works:** https://github.com/morrownr/rtw89 — the rtw89 driver from kernel 6.14 backported to build on 6.6+. On my 6.8.0 kernel:

```
sudo apt remove bcmwl-kernel-source   # if present; wrongly matches this chip
git clone https://github.com/morrownr/rtw89
cd rtw89
make
sudo make install
sudo make install_fw
sudo cp rtw89.conf /etc/modprobe.d/
sudo dkms install .                   # optional, survives kernel updates
sudo modprobe rtw89_8922ae_git
```

Probe is clean (`Firmware version 0.35.80.3`, `CV: 2` recognized), scan/connect/WPA3 all working.