lwfinger/rtw89 #386: kernel log full of "PHY c2h class 2 not support" messages

Sfinx, 2025-05-11:
```
[May11 06:48] rtw89_8922ae 0000:01:00.0: PHY c2h class 2 not support
[  +0.047258] rtw89_8922ae 0000:01:00.0: PHY c2h class 2 not support
[  +0.046598] rtw89_8922ae 0000:01:00.0: PHY c2h class 2 not support
[  +0.054118] rtw89_8922ae 0000:01:00.0: PHY c2h class 2 not support
[  +0.047014] rtw89_8922ae 0000:01:00.0: PHY c2h class 2 not support
[  +0.046638] rtw89_8922ae 0000:01:00.0: PHY c2h class 2 not support
[  +0.053837] rtw89_8922ae 0000:01:00.0: PHY c2h class 2 not support
[  +0.047245] rtw89_8922ae 0000:01:00.0: PHY c2h class 2 not support
...
```

x86_64 kernel v6.14.6, h/w Realtek Semiconductor Co., Ltd. Device 8922 (rev 01)

a5a5aa555oo, 2025-05-24:
Hi, 

Could you post the full output of dmesg?

```
sudo dmesg
```

Issif, 2025-05-24:
Hello,

I'm facing the same issue:

```
[64597.482470] rtw89_8922ae 0000:03:00.0: PHY c2h class 2 not support
[64597.590914] rtw89_8922ae 0000:03:00.0: PHY c2h class 2 not support
[64597.699323] rtw89_8922ae 0000:03:00.0: PHY c2h class 2 not support
[64597.807713] rtw89_8922ae 0000:03:00.0: PHY c2h class 2 not support
[64598.066031] wlo1: authenticate with 38:b5:c9:0d:9a:65 (local address=70:08:94:c0:0a:85)
[64598.066044] wlo1: send auth to 38:b5:c9:0d:9a:65 (try 1/3)
[64598.067042] wlo1: authenticated
[64598.067323] wlo1: associate with 38:b5:c9:0d:9a:65 (try 1/3)
[64598.070625] wlo1: RX AssocResp from 38:b5:c9:0d:9a:65 (capab=0x1511 status=0 aid=92)
[64598.176420] wlo1: associated
[64598.176491] wlo1: Limiting TX power to 23 (23 - 0) dBm as advertised by 38:b5:c9:0d:9a:65
[64613.582699] rtw89_8922ae 0000:03:00.0: failed to update 168 RXBD info: -11
[64613.582720] rtw89_8922ae 0000:03:00.0: failed to update 169 RXBD info: -11
[64636.490953] rtw89_8922ae 0000:03:00.0: failed to update 238 RXBD info: -11
[64636.491003] rtw89_8922ae 0000:03:00.0: failed to update 239 RXBD info: -11
[64673.072968] wlo1: Connection to AP 38:b5:c9:0d:9a:65 lost
[64673.371342] rtw89_8922ae 0000:03:00.0: PHY c2h class 2 not support
[64673.564233] wlo1: authenticate with 38:b5:c9:0d:9a:60 (local address=70:08:94:c0:0a:85)
[64673.564244] wlo1: send auth to 38:b5:c9:0d:9a:60 (try 1/3)
[64673.567330] wlo1: authenticated
[64673.567779] wlo1: associate with 38:b5:c9:0d:9a:60 (try 1/3)
[64673.575717] wlo1: RX AssocResp from 38:b5:c9:0d:9a:60 (capab=0x1411 status=0 aid=1)
[64673.684926] wlo1: associated
[64673.933608] wlo1: Limiting TX power to 20 (20 - 0) dBm as advertised by 38:b5:c9:0d:9a:60
[64790.477230] rtw89_8922ae 0000:03:00.0: PHY c2h class 2 not support
[64790.524490] rtw89_8922ae 0000:03:00.0: PHY c2h class 2 not support
[64790.571031] rtw89_8922ae 0000:03:00.0: PHY c2h class 2 not support
[64790.625180] rtw89_8922ae 0000:03:00.0: PHY c2h class 2 not support
[64790.672311] rtw89_8922ae 0000:03:00.0: PHY c2h class 2 not support
[64790.718814] rtw89_8922ae 0000:03:00.0: PHY c2h class 2 not support
```

OS: `Manjaro Linux 25.0.3`
Kernel: `6.14.6-2-MANJARO #1 SMP PREEMPT_DYNAMIC Wed, 14 May 2025 14:56:34 +0000 x86_64 GNU/Linux`
Network Controller + Driver:
```
03:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8922AE 802.11be PCIe Wireless Network Adapter (rev 01)
        DeviceName: Realtek Ethernet
        Subsystem: Lenovo Device 4922
        Kernel driver in use: rtw89_8922ae
        Kernel modules: rtw89_8922ae
```

a5a5aa555oo, 2025-05-24:
@Issif 

Please see this [reply](https://github.com/lwfinger/rtw89/issues/382#issuecomment-2848650836).

Or try this [rtw89 driver](https://github.com/morrownr/rtw89) that will be in kernel 6.16.

Issif, 2025-05-24:
> [@Issif](https://github.com/Issif)
> 
> Please see this [reply](https://github.com/lwfinger/rtw89/issues/382#issuecomment-2848650836).
> 
> Or try this [rtw89 driver](https://github.com/morrownr/rtw89?rgh-link-date=2025-05-24T09%3A02%3A32.000Z) that will be in kernel 6.16.

I compiled and installed the rtw89 driver from the main branch, I'll see if the situation is better and I'll let you know here in a couple of days.

Thanks for your help

yujezhang, 2025-06-14:
> [@Issif](https://github.com/Issif)
> 
> Please see this [reply](https://github.com/lwfinger/rtw89/issues/382#issuecomment-2848650836).
> 
> Or try this [rtw89 driver](https://github.com/morrownr/rtw89) that will be in kernel 6.16.

This works perfectly

tectrixdev, 2025-08-14:
> Or try this [rtw89 driver](https://github.com/morrownr/rtw89) that will be in kernel 6.16.

Hi, you're saying it will be in kernel 6.16. It's out now. How can we verify that this is merged? I really want to know because I can't install the driver because of RHEL(fedora). Is it already merged?

tectrixdev, 2025-08-14:
> [@tectrixdev](https://github.com/tectrixdev)
> 
> Which version of RHEL are you running? kernel version? what Wi-Fi card/adapter are you using?
> 
> Currently, the rtw89 driver in morrownr's repo = the rtw89 driver in kernel 6.17 + additional drivers for RTL8852CU/RTL8912AU/RTL8922AU written by Bitterblue Smith ([@dubhater](https://github.com/dubhater))
> 
> [kernel 6.17-rc1](https://git.kernel.org/torvalds/t/linux-6.17-rc1.tar.gz) is out, you can use [meld](https://meldmerge.org/) or other tools to verify.

I'm using the RTL8922AE wifi card, Fedora KDE 42, kernel: Linux 6.15.9-201.fc42.x86_64.

So if I'm right, the kernel driver is implemented in 6.17? So I will receive it when fedora updates their kernel version?

a5a5aa555oo, 2025-08-14:
@tectrixdev 

The rtw89 driver has been merged into the kernel since [5.16](https://github.com/torvalds/linux/commit/e3ec7017f6a20d12ddd9fe23d345ebb7b8c104dd).

> So I will receive it when fedora updates their kernel version?

Yes!

a5a5aa555oo, 2025-08-14:
> This one should fix that right?

If the rtw89 driver in your kernel 6.15 doesn't work well, you can try the one in morrownr's repo because it is newer and may already fix the problems you encountered.

btw, i dont have rtl8922ae card.

tectrixdev, 2025-08-14:
> > Morrownr's driver is the one merged in 6.17, correct? If so I should just wait until fedora ships it right?
> 
> It's up to you. If the problems in the rtw89 driver you are using is tolerable, then you can wait.
> 
> UPDATE: Sorry I misunderstood what you said. You can try morrownr's driver right now, it is compatible with kernel 6.15.

It's compatible, I just doesn't function on RHEL based OS because they apparently made changes to the kernel API, it's said in the readme of the project, I tried, driver loads but doesn't function.

a5a5aa555oo, 2025-08-14:
@tectrixdev 

Open an issue there and attach the full output of dmesg.
```
sudo dmesg > dmesg.txt
```

tectrixdev, 2025-08-14:
I got it to work, @a5a5aa555oo When I get kernel 6.17, will the module just unload itself or do I have to uninstall it?

a5a5aa555oo, 2025-08-14:
@tectrixdev 

You have to uninstall it by yourself, it is not that smart. :)

tectrixdev, 2025-08-14:
What if I would leave it? Will it just override kernel module still?

a5a5aa555oo, 2025-08-14:
> Will it just override kernel module still

Yes, the lines starting with `blacklist` in /etc/modprobe.d/[rtw89.conf](https://github.com/morrownr/rtw89/blob/main/rtw89.conf) will prevent the in-kernel rtw89 driver from loading.

Sfinx, 2025-08-15:
I do not see this error anymore in stock 6.16

closed 2025-08-15 by Sfinx