lwfinger/rtw89 #388: Error running make

frownsty, 2025-06-06:
in os: Linux parrot 6.12.12-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.12.12-1parrot1 (2025-02-27) x86_64 GNU/Linux

ran make in the rtw89 directory, error:

make -C /lib/modules/6.12.12-amd64/build M=/home/dolphin/rtw89 modules
make[1]: Entering directory '/usr/src/linux-headers-6.12.12-amd64'
  CC [M]  /home/dolphin/rtw89/mac80211.o
/home/dolphin/rtw89/mac80211.c:1177:35: error: initialization of ‘void (*)(struct ieee80211_hw *, bool)’ {aka ‘void (*)(struct ieee80211_hw *, _Bool)’} from incompatible pointer type ‘void (*)(struct ieee80211_hw *)’ [-Werror=incompatible-pointer-types]
 1177 |         .stop                   = rtw89_ops_stop,
      |                                   ^~~~~~~~~~~~~~
/home/dolphin/rtw89/mac80211.c:1177:35: note: (near initialization for ‘rtw89_ops.stop’)
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-6.12.12-common/scripts/Makefile.build:234: /home/dolphin/rtw89/mac80211.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.12.12-common/Makefile:1962: /home/dolphin/rtw89] Error 2
make[1]: *** [/usr/src/linux-headers-6.12.12-common/Makefile:236: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.12.12-amd64'

Trying to install the 8852ce driver

a5a5aa555oo, 2025-06-06:
Could you post the output of this command?
```
sudo dmesg
```

frownsty, 2025-06-06:
dmesg grepped for rtw, since there is a bunch of other junk in there about my usb tethering connection

`
[    9.480833] rtw89_8852ce 0000:02:00.0: loaded firmware rtw89/rtw8852c_fw-1.bin
[    9.481248] rtw89_8852ce 0000:02:00.0: enabling device (0000 -> 0003)
[    9.482327] rtw89_8852ce 0000:02:00.0: Firmware version 0.27.125.0 (526f84c0), cmd version 0, type 1
[    9.482330] rtw89_8852ce 0000:02:00.0: Firmware version 0.27.125.0 (526f84c0), cmd version 0, type 3
[    9.486575] rtw89_8852ce 0000:02:00.0: [ERR]fw bin size
[    9.486581] rtw89_8852ce 0000:02:00.0: parse fw header fail
[    9.486584] rtw89_8852ce 0000:02:00.0: [ERR]fwdl 0x1E0 = 0x1
[    9.486587] rtw89_8852ce 0000:02:00.0: [ERR]fwdl 0x83F0 = 0xdeadbeef
[    9.486594] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002be
[    9.486607] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486620] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486633] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486645] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486658] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b0
[    9.486671] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002aa
[    9.486684] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486696] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486709] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486722] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486735] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486747] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486760] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486773] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002be
[    9.486814] rtw89_8852ce 0000:02:00.0: [ERR]fw bin size
[    9.486815] rtw89_8852ce 0000:02:00.0: parse fw header fail
[    9.486818] rtw89_8852ce 0000:02:00.0: [ERR]fwdl 0x1E0 = 0x1
[    9.486820] rtw89_8852ce 0000:02:00.0: [ERR]fwdl 0x83F0 = 0xdeadbeef
[    9.486826] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486839] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486851] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486864] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486879] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486891] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486904] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486916] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486929] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486943] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002be
[    9.486955] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486968] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.486982] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b4
[    9.486995] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487007] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487061] rtw89_8852ce 0000:02:00.0: [ERR]fw bin size
[    9.487062] rtw89_8852ce 0000:02:00.0: parse fw header fail
[    9.487065] rtw89_8852ce 0000:02:00.0: [ERR]fwdl 0x1E0 = 0x1
[    9.487067] rtw89_8852ce 0000:02:00.0: [ERR]fwdl 0x83F0 = 0xdeadbeef
[    9.487074] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487089] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487102] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487115] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487130] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487142] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b2
[    9.487155] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487170] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b0
[    9.487183] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487195] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487208] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b0
[    9.487221] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b2
[    9.487233] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487246] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487258] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002be
[    9.487300] rtw89_8852ce 0000:02:00.0: [ERR]fw bin size
[    9.487301] rtw89_8852ce 0000:02:00.0: parse fw header fail
[    9.487303] rtw89_8852ce 0000:02:00.0: [ERR]fwdl 0x1E0 = 0x1
[    9.487306] rtw89_8852ce 0000:02:00.0: [ERR]fwdl 0x83F0 = 0xdeadbeef
[    9.487311] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487324] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[... 14 lines cut ...]
[    9.487531] rtw89_8852ce 0000:02:00.0: parse fw header fail
[    9.487533] rtw89_8852ce 0000:02:00.0: [ERR]fwdl 0x1E0 = 0x1
[    9.487535] rtw89_8852ce 0000:02:00.0: [ERR]fwdl 0x83F0 = 0xdeadbeef
[    9.487541] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b4
[    9.487554] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b4
[    9.487566] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487579] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487592] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487606] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487619] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[    9.487634] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b4
[    9.487647] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002aa
[    9.487661] rtw89_8852ce 0000:02:00.0: [ERR]fw PC = 0x200002b6
[... cut ...] hda_codec intel_uncore_frequency uvcvideo intel_uncore_frequency_common x86_pkg_temp_thermal videobuf2_vmalloc intel_powerclamp snd_hda_core uvc coretemp dell_pc snd_hwdep videobuf2_memops kvm_intel platform_profile mac80211 snd_pcm mei_pxp videobuf2_v4l2 mei_hdcp kvm btusb videodev nls_cp437 processor_thermal_device_pci btrtl snd_timer dell_laptop iTCO_wdt btintel intel_pmc_bxt videobuf2_common dell_wmi vfat processor_thermal_device btbcm libarc4

`

I will try the 6.16 one real quick thank you

a5a5aa555oo, 2025-06-06:
The firmware for 8852CE provided by your distro is too new so that [the driver is unable to parse it](https://web.git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/commit/rtw89/rtw8852c_fw-1.bin?id=cb465aa0042bc2057579b313f518a80a2207a55c).

There are two solutions available for you:
1. Downgrade the firmware to v0.27.122.0.
2. Install the rtw89 driver maintained by morrownr.

The second one would be easier.

frownsty, 2025-06-06:
Nothing really stands out to me:

[   12.752302] rtw89_8852ce_git 0000:02:00.0: loaded firmware rtw89/rtw8852c_fw-2.bin
[   12.757599] rtw89_8852ce_git 0000:02:00.0: enabling device (0000 -> 0003)
[   12.759202] rtw89_8852ce_git 0000:02:00.0: MAC has already powered on
[   12.768833] rtw89_8852ce_git 0000:02:00.0: Firmware version 0.27.125.0 (526f84c0), cmd version 0, type 1
[   12.768837] rtw89_8852ce_git 0000:02:00.0: Firmware version 0.27.125.0 (526f84c0), cmd version 0, type 3
[   13.058596] rtw89_8852ce_git 0000:02:00.0: chip rfe_type is 1
[   13.083602] rtw89_8852ce_git 0000:02:00.0: Firmware element BB version: 00 28 00 00
[   13.083628] rtw89_8852ce_git 0000:02:00.0: Firmware element radio A version: 00 63 00 00
[   13.083645] rtw89_8852ce_git 0000:02:00.0: Firmware element NCTL version: 00 11 00 00
[   13.083701] rtw89_8852ce_git 0000:02:00.0: Firmware element TXPWR version: 00 78 00 00
[   13.083703] rtw89_8852ce_git 0000:02:00.0: Firmware element PWR_TRK version: 00 26 00 00
[   13.090950] rtw89_8852ce_git 0000:02:00.0: rfkill hardware state changed to enable
[   13.128152] rtw89_8852ce_git 0000:02:00.0 wlp2s0: renamed from wlan0
[   17.650154] wlp2s0: authenticate with b8:f8:53:61:97:0a (local address=84:9e:56:9a:39:a7)
[   17.650159] wlp2s0: send auth to b8:f8:53:61:97:0a (try 1/3)
[   17.652058] wlp2s0: authenticated
[   17.653262] wlp2s0: associate with b8:f8:53:61:97:0a (try 1/3)
[   17.662403] wlp2s0: RX AssocResp from b8:f8:53:61:97:0a (capab=0x1411 status=0 aid=12)
[   17.774290] wlp2s0: associated
[   17.774406] wlp2s0: Limiting TX power to 30 (30 - 0) dBm as advertised by b8:f8:53:61:97:0a
[   95.120224] wlp2s0: deauthenticating from b8:f8:53:61:97:0a by local choice (Reason: 3=DEAUTH_LEAVING)
[   99.439795] wlp2s0: authenticate with b8:f8:53:61:97:0a (local address=84:9e:56:9a:39:a7)
[   99.439800] wlp2s0: send auth to b8:f8:53:61:97:0a (try 1/3)
[   99.448170] wlp2s0: authenticated
[   99.450005] wlp2s0: associate with b8:f8:53:61:97:0a (try 1/3)
[   99.457746] wlp2s0: RX AssocResp from b8:f8:53:61:97:0a (capab=0x1411 status=0 aid=15)
[   99.567106] wlp2s0: associated
[   99.621661] wlp2s0: Limiting TX power to 30 (30 - 0) dBm as advertised by b8:f8:53:61:97:0a
[  155.747746] wlp2s0: deauthenticating from b8:f8:53:61:97:0a by local choice (Reason: 3=DEAUTH_LEAVING)
[  174.496622] wlp2s0: authenticate with e6:ee:c7:55:1b:10 (local address=84:9e:56:9a:39:a7)
[  174.496626] wlp2s0: send auth to e6:ee:c7:55:1b:10 (try 1/3)
[  174.502629] wlp2s0: authenticated
[  174.503083] wlp2s0: associate with e6:ee:c7:55:1b:10 (try 1/3)
[  174.555937] wlp2s0: RX AssocResp from e6:ee:c7:55:1b:10 (capab=0x1431 status=0 aid=1)
[  174.667218] wlp2s0: associated
[  174.667327] wlp2s0: Limiting TX power to 30 (30 - 0) dBm as advertised by e6:ee:c7:55:1b:10
[  201.576051] wlp2s0: deauthenticating from e6:ee:c7:55:1b:10 by local choice (Reason: 3=DEAUTH_LEAVING)
[  205.036189] wlp2s0: authenticate with b8:f8:53:61:97:0a (local address=84:9e:56:9a:39:a7)
[  205.036192] wlp2s0: send auth to b8:f8:53:61:97:0a (try 1/3)
[  205.038998] wlp2s0: authenticated
[  205.039784] wlp2s0: associate with b8:f8:53:61:97:0a (try 1/3)
[  205.044681] wlp2s0: RX AssocResp from b8:f8:53:61:97:0a (capab=0x1411 status=0 aid=17)
[  205.153841] wlp2s0: associated
[  205.154011] wlp2s0: Limiting TX power to 30 (30 - 0) dBm as advertised by b8:f8:53:61:97:0a

a5a5aa555oo, 2025-06-07:
just installed Parrot Linux in a VM, tested morrownr's rtw89 driver with 8852BE and it works..no idea why your 8852CE keeps disconnecting.

Do you wanna try the first solution?

Steps:

1. Completely remove the rtw89 driver that comes from morrownr's repo.
```
sudo dkms remove rtw89/6.16 --all
```
```
sudo rm -rf /usr/src/rtw89*
```
```
sudo rm -f /etc/modprobe.d/rtw89.conf
```

2. Downgrade firmware for 8852CE to v0.27.122.0.
```
git clone https://kernel.googlesource.com/pub/scm/linux/kernel/git/firmware/linux-firmware.git
```
```
cd linux-firmware
```
```
git reset --hard b93e274c
```
```
# copy firmware v0.27.122.0 to /lib/firmware/rtw89
sudo cp rtw89/rtw8852c_fw-1.bin /lib/firmware/rtw89
```

After reboot, the rtw89 in kernel 6.12 will be loaded and it will try to load the older firmware v0.27.122.0.

a5a5aa555oo, 2025-06-07:
morrownr's rtw89 driver was not removed completely.

How did you install morrown's rtw89 driver? via `make` or `dkms`? please follow [this guide](https://github.com/morrownr/rtw89?tab=readme-ov-file#uninstallation-guide) to remove it again.

frownsty, 2025-06-07:
I had it installed with both, which is why it didn't uninstall properly when I did it the first time. New dmesg:

`
[   10.413267] rtw89_8852ce 0000:02:00.0: loaded firmware rtw89/rtw8852c_fw-1.bin
[   10.413380] rtw89_8852ce 0000:02:00.0: enabling device (0000 -> 0003)
[   10.414320] rtw89_8852ce 0000:02:00.0: Firmware version 0.27.122.0 (8d4a0ef2), cmd version 0, type 1
[   10.414322] rtw89_8852ce 0000:02:00.0: Firmware version 0.27.122.0 (8d4a0ef2), cmd version 0, type 3
[   10.704570] rtw89_8852ce 0000:02:00.0: chip rfe_type is 1
[   10.738456] rtw89_8852ce 0000:02:00.0: Firmware element BB version: 00 28 00 00
[   10.738525] rtw89_8852ce 0000:02:00.0: Firmware element radio A version: 00 63 00 00
[   10.738573] rtw89_8852ce 0000:02:00.0: Firmware element NCTL version: 00 11 00 00
[   10.738643] rtw89_8852ce 0000:02:00.0: Firmware element TXPWR version: 00 69 00 00
[   10.738647] rtw89_8852ce 0000:02:00.0: Firmware element PWR_TRK version: 00 26 00 00
[   10.742678] rtw89_8852ce 0000:02:00.0: rfkill hardware state changed to enable
[   10.794277] Modules linked in: wl(POE+) bnep binfmt_misc snd_sof_pci_intel_tgl snd_sof_pci_intel_cnl snd_sof_intel_hda_generic soundwire_intel soundwire_generic_allocation rtw89_8852ce soundwire_cadence snd_sof_intel_hda_common rtw89_8852c snd_soc_hdac_hda snd_sof_intel_hda_mlink snd_sof_intel_hda snd_sof_pci snd_sof_xtensa_dsp snd_sof rtw89_pci snd_sof_utils rtw89_core snd_soc_acpi_intel_match snd_soc_acpi soundwire_bus snd_ctl_led snd_soc_avs snd_hda_codec_realtek(+) snd_soc_hda_codec snd_hda_codec_generic snd_hda_ext_core dell_pc platform_profile snd_hda_scodec_component snd_soc_core mac80211 snd_compress snd_pcm_dmaengine uvcvideo snd_hda_intel snd_intel_dspcfg snd_intel_sdw_acpi snd_hda_codec videobuf2_vmalloc uvc videobuf2_memops intel_uncore_frequency videobuf2_v4l2 intel_uncore_frequency_common x86_pkg_temp_thermal snd_hda_core intel_powerclamp snd_hwdep videodev btusb coretemp snd_pcm btrtl kvm_intel btintel dell_laptop mei_hdcp videobuf2_common btbcm mei_pxp iTCO_wdt libarc4 intel_rapl_msr kvm btmtk rapl
[   10.795820] rtw89_8852ce 0000:02:00.0 wlp2s0: renamed from wlan0
[   13.207637] rtw89_8852ce 0000:02:00.0: c2h class 9 func 6 not support
[   17.115171] rtw89_8852ce 0000:02:00.0: c2h class 9 func 6 not support
[   19.117996] rtw89_8852ce 0000:02:00.0: c2h class 9 func 6 not support
[   21.122162] rtw89_8852ce 0000:02:00.0: c2h class 9 func 6 not support
[   23.123588] rtw89_8852ce 0000:02:00.0: c2h class 9 func 6 not support
[   25.127008] rtw89_8852ce 0000:02:00.0: c2h class 9 func 6 not support
[   31.246591] rtw89_8852ce 0000:02:00.0: c2h class 9 func 6 not support
[   34.665080] rtw89_8852ce 0000:02:00.0: c2h class 9 func 6 not support
[   41.064308] rtw89_8852ce 0000:02:00.0: c2h class 9 func 6 not support
[   43.067127] rtw89_8852ce 0000:02:00.0: c2h class 9 func 6 not support
[   45.069878] rtw89_8852ce 0000:02:00.0: c2h class 9 func 6 not support
[   47.072845] rtw89_8852ce 0000:02:00.0: c2h class 9 func 6 not support
[   49.265599] rtw89_8852ce 0000:02:00.0: c2h class 9 func 6 not support
[   51.268512] rtw89_8852ce 0000:02:00.0: c2h class 9 func 6 not support
[   53.271490] rtw89_8852ce 0000:02:00.0: c2h class 9 func 6 not support
[   55.274271] rtw89_8852ce 0000:02:00.0: c2h class 9 func 6 not support
`

frownsty, 2025-06-07:
This solved it for me, thank you

closed 2025-06-07 by frownsty