lwfinger/rtw89 #396: [6.15.8] error: implicit declaration of function ‘init_dummy_netdev’

alexisfrjp, 2025-07-30:
```
$ make
make -C /lib/modules/6.15.8-1.el9.elrepo.x86_64/build M=/home/alexis/git/rtw89 modules
make[1]: Entering directory '/usr/src/kernels/6.15.8-1.el9.elrepo.x86_64'
make[2]: Entering directory '/home/alexis/git/rtw89'
  CC [M]  core.o
core.c: In function ‘rtw89_core_napi_init’:
core.c:2591:9: error: implicit declaration of function ‘init_dummy_netdev’ [-Werror=implicit-function-declaration]
 2591 |         init_dummy_netdev(&rtwdev->netdev);
      |         ^~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/kernels/6.15.8-1.el9.elrepo.x86_64/scripts/Makefile.build:203: core.o] Error 1
make[3]: *** [/usr/src/kernels/6.15.8-1.el9.elrepo.x86_64/Makefile:2003: .] Error 2
make[2]: *** [/usr/src/kernels/6.15.8-1.el9.elrepo.x86_64/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/home/alexis/git/rtw89'
make[1]: *** [Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.15.8-1.el9.elrepo.x86_64'
make: *** [Makefile:104: all] Error 2

```

a5a5aa555oo, 2025-08-10:
[The Linux Community Lost Wireless Pioneer Larry Finger](https://salvacybersec.medium.com/the-linux-community-lost-wireless-pioneer-larry-finger-ff7efd415da7)

The rtw89 driver in this repo = the rtw89 driver in the kernel 6.10

Doesn't the rtw89 driver in your kernel work? why were you trying to install an older driver? 😩😩😩

a5a5aa555oo, 2025-08-10:
what chip do you have? any helpful messages in the kernel log?

`sudo dmesg | grep rtw89`

alexisfrjp, 2025-08-10:
I have a `Realtek Wi-Fi 6 RTL8851BE` from motherboard `Gigabyte H810M GAMING WIFI6`.
Can't access it until tomorrow but last time nothing was shown for wifi with the 6.15.8 kernel. Bluetooth worked tho.

a5a5aa555oo, 2025-08-14:
@alexisfrjp 

Try the driver here. https://github.com/morrownr/rtw89

Open an issue there if it doesn't work, either.

a5a5aa555oo, 2025-11-17:
I may know what is wrong. 

The RTL8852BE card has a compatibility problem with Intel X99 chipset and a hacked rtw89 driver is needed to make it work.

Please

1.  Follow the [Uninstallation Guide](https://github.com/morrownr/rtw89?tab=readme-ov-file#uninstallation-guide) to remove the rtw89 driver installed earlier. 

2. Follow the [Installation Guide](https://github.com/a5a5aa555oo/rtw89-1?tab=readme-ov-file#installation-guide) to install the hacked rtw89 driver in my repo.

```
git clone https://github.com/a5a5aa555oo/rtw89-1
```
```
cd rtw89-1
```
```
sudo dkms install $PWD
```
```
sudo make install_fw
```
```
sudo cp -v rtw89.conf /etc/modprobe.d/
```
The last step is important, please must copy the configuration file `rtw89.conf` to `/etc/modprobe.d`.

Attach the kernel log again if your RTL8852BE card still doesn't work.

iohann95, 2025-11-17:
> I may know what is wrong.
>
> The RTL8852BE card has a compatibility problem with Intel X99 chipset and a hacked rtw89 driver is needed to make it work.
>
> Please
>
>     1. Follow the [Uninstallation [...]

[a5a5aa555oo](https://github.com/a5a5aa555oo) It worked perfectly, I cannot thank you enough.
I can finally ditch Windows as my daily driver.

Unfortunately wifi boards with intel chips were too expensive and also I'm stuck with X99 as parts are too expensive lately.

Thanks!!!!!!!!

a5a5aa555oo, 2026-01-29:
@iohann95 Thank you! Let me know if you need more info.

iohann95, 2026-01-31:
[lspci_x.txt](https://github.com/user-attachments/files/24976664/lspci_x.txt)
[lspci_vt.txt](https://github.com/user-attachments/files/24976666/lspci_vt.txt)
[dmidecode.txt](https://github.com/user-attachments/files/24976667/dmidecode.txt)

@a5a5aa555oo , @ulli-kroll , Attaching here my environment details, maybe this can help. 👍

iohann95, 2026-10-15:
> [@iohann95](https://github.com/iohann95)
> 
> I've merged Ping-Ke's fix into morrownr's rtw89 repo, you can
> 
>     1. [Completely remove the rtw89 driver installed.](https://github.com/a5a5aa555oo/rtw89-1?tab=readme-ov-file#uninstallation-guide) Ensure `/etc/modprobe.d/rtw89.conf` is removed. Otherwise the in-kernel rtw89 will not be loaded on next boot.
> 
>     2. Reboot to kernel 6.14 to reproduce the problem. It would be good if you can attatch the kernel log so we can confirm what problem it exactly it is again. (`sudo dmesg > dmesg_6.14.txt `)
> 
>     3. [Install the rtw89 driver from morrownr's repo](https://github.com/morrownr/rtw89?tab=readme-ov-file#installation-guide)
> 
>     4. Reboot to see if the problem is fixed. Also good if you can attatch the kernel log again. (`sudo dmesg > dmesg_morrownr.txt `)

Good evening,
Unfortunately it did not work... I kind of gave up as I had several other issues with this machine with any kind of Linux flavor. It only works 100% on Windows. I'll use ubuntu on my laptop only.

but thanks!